Semi-ATE/Semi-ATE #12: Handle exceptions

goanpeca, 2020-07-27:
Hi @nerohmot with [this pr](https://github.com/spyder-ide/spyder/pull/13375) merged, you can report errors using the Spyder builtin report dialog from either the plugin or the main_widget.

If the exception happens on a child widget different from the main_widget, just bubble a signal up with the needed information.

This is how it can be used

```python

    sig_exception_occurred = Signal(dict)
    """
    This signal can be emitted to report an exception from any plugin.
    Parameters
    ----------
    error_data: dict
        The dictionary containing error data. The expected keys are:
        >>> error_data= {
            "text": str,
            "is_traceback": bool,
            "repo": str,
            "title": str,
            "label": str,
            "steps": str,
        }
    Notes
    -----
    The `is_traceback` key indicates if `text` contains plain text or a
    Python error traceback.
    The `title` and `repo` keys indicate how the error data should
    customize the report dialog and Github error submission.
    The `label` and `steps` keys allow customizing the content of the
    error dialog.
    This signal is automatically connected to the main container/widget.
    """
```

nerohmot, 2020-07-28:
but we need to have #13375 form spyder right?

goanpeca, 2020-07-28:
Correct!

nerohmot, 2020-08-02:
Ok, so now we also have [#13375](https://github.com/spyder-ide/spyder/pull/13375) in master :+1:

awinia-github, 2022-06-21:
old issues

closed 2022-06-21 by awinia-github