Semi-ATE/Semi-ATE #97: If generated testfile running under spyder, no stdf-result file will be created.

Zlin526F, 2021-05-12:
The launch_master.py and the python launch_control.py isn't running under spyder:

-- Error-Message launch_master.py:
  File "....\Anaconda3\envs\Semi-ATE1\lib\asyncio\base_events.py", line 552, in _check_running
    raise RuntimeError('This event loop is already running')
RuntimeError: This event loop is already running

-- Error-Message launch_control.py:
  File "C:\Users\jung\Anaconda3\envs\Semi-ATE1\lib\asyncio\runners.py", line 33, in run
    raise RuntimeError(
RuntimeError: asyncio.run() cannot be called from a running event loop

Workaround: Start direct the generated testfile  (e.q. tb_ate_HW0_FT_Device1_engineering_1.py).
But than, no stdf-result file will be created.
For this, we need a master_light application.

awinia-github, 2021-05-17:
SPYD-481

Zlin526F, 2021-08-04:
ok,with the pull request 103 , stdf will create directly with the spyder environment :-)

closed 2021-08-04 by Zlin526F