Semi-ATE/Semi-ATE #111: after first installation and open an existing project, the target combo box is empty

Zlin526F, 2021-11-15:
after first installation and open an existing project, the target combo box is empty
[image]

You have to restart Spyder  and than the target is ok
[image]
:

awinia-github, 2022-06-21:
not reproducible

Zlin526F, 2022-06-22:
This error occure, if I create a new mamba environment,  install spyder=5.3 and the latest semi-ate. After starting spyder and opening an existing project, set the Hardware and Base Comboxes, than the target box is empty. I have to restart spyder. 
Only then do I have a selection in the target.

closed 2022-07-04 by awinia-github