Semi-ATE/Semi-ATE #129: sammy -h -> get an exception

Zlin526F, 2022-02-07:
[image]

nerohmot, 2022-02-07:
... you see now why I insist on the packaging with Awinia?

Zlin526F, 2022-02-07:
yes, sometimes it really hurts.

I try to get an existing tb_ate running again with sammy, but with 'sammy generate test_target' i get an exception, too:
[image]

Zlin526F, 2022-02-07:
and yet another error
sammy build isn' t running together with maxiconda:
sammy.ps1: line 10:
else
{
    Invoke-Expression "conda create --name sammybuildn"
    Invoke-Expression "conda activate sammybuildn"
    Invoke-Expression "conda install --file run.txt"
    Invoke-Expression "pyinstaller --onefile --console --clean --distpath .\ sammy.py"
    Invoke-Expression "conda deactivate"
}

-> change conda to mamba

nerohmot, 2022-02-07:
Yes, that should be "mamba create -n sammybuildn" !

Zlin526F, 2022-02-07:
By incorrect deletion, check-in and updates with svn,  the database can be destroyed.
To find this errors in the database, it will be helpful to add in ATE\sammy\verbs\generate.py in line 133:
            try:
                testdefinition = Test.get(self.file_operator, test_target.test, test_target.hardware, test_target.base).definition
            except Exception:
               print(f"        Error: couldn't generate {test_target.hardware}, {test_target.base}, {test_target.test}") 
               continue
--> no exception occure, only an Error. With the information you can repair the database by hand.

Zlin526F, 2022-04-08:
ok in Version 1.03

closed 2022-04-08 by Zlin526F