Semi-ATE/Semi-ATE #168: The spyder breakpoints doesn't work

Zlin526F, 2022-06-02:
If you set a spyder breakpoint in a test and start the testprogramm in debug mode, the breakpoint doesn't stop the running code.
You have to set a breakpoint() (write it in your code) once in a test program.

# Mandatory:
atomicwrites >=1.2.0          :  1.4.0 (OK)
chardet >=2.0.0               :  4.0.0 (OK)
cloudpickle >=0.5.0           :  2.1.0 (OK)
cookiecutter >=1.6.0          :  1.7.3 (OK)
diff_match_patch >=20181111   :  20200713 (OK)
intervaltree >=3.0.2          :  3.0.2 (OK)
IPython >=7.31.1;<8.0.0       :  7.33.0 (OK)
jedi >=0.17.2;<0.19.0         :  0.18.1 (OK)
jellyfish >=0.7               :  0.9.0 (OK)
jsonschema >=3.2.0            :  4.5.1 (OK)
keyring >=17.0.0              :  23.5.1 (OK)
nbconvert >=4.0               :  6.5.0 (OK)
numpydoc >=0.6.0              :  1.2.1 (OK)
paramiko >=2.4.0              :  2.11.0 (OK)
parso >=0.7.0;<0.9.0          :  0.8.3 (OK)
pexpect >=4.4.0               :  4.8.0 (OK)
pickleshare >=0.4             :  0.7.5 (OK)
psutil >=5.3                  :  5.9.1 (OK)
pygments >=2.0                :  2.12.0 (OK)
pylint >=2.5.0                :  2.13.9 (OK)
pyls_spyder >=0.4.0           :  0.4.0 (OK)
pylsp >=1.4.1;<1.5.0          :  1.4.1 (OK)
pylsp_black >=1.2.0           :  1.2.1 (OK)
qdarkstyle >=3.0.2;<3.1.0     :  3.0.3 (OK)
qstylizer >=0.1.10            :  0.2.1 (OK)
qtawesome >=1.0.2             :  1.1.1 (OK)
qtconsole >=5.3.0;<5.4.0      :  5.3.0 (OK)
qtpy >=2.1.0                  :  2.1.0 (OK)
rtree >=0.9.7                 :  1.0.0 (OK)
setuptools >=49.6.0           :  62.3.2 (OK)
sphinx >=0.6.6                :  4.5.0 (OK)
spyder_kernels >=2.3.1;<2.4.0 :  2.3.1 (OK)
textdistance >=4.2.0          :  4.2.2 (OK)
three_merge >=0.1.1           :  0.1.1 (OK)
watchdog >=0.10.3             :  2.1.8 (OK)
zmq >=22.1.0                  :  23.0.0 (OK)

# Optional:
cython >=0.21                 :  None (NOK)
matplotlib >=3.0.0            :  3.5.2 (OK)
numpy >=1.7                   :  1.22.4 (OK)
pandas >=1.1.1                :  1.4.2 (OK)
scipy >=0.17.0                :  None (NOK)
sympy >=0.7.3                 :  None (NOK)

# Spyder plugins:
ate_spyder.plugin 1.0.9       :  1.0.9 (OK)

[image]

Zlin526F, 2022-07-07:
@steff456  : to reproduce the error you have to run a complete project of Semi-ATE, but unfortunately you still have to install a lot :-(
1. Install all packages from semi-ate
2. You need an mqtt-broker on your machine,
     - mamba install mosquitto
     - mosquitto -d
     - or https://mosquitto.org/download/  (if you use Windows, you have to restart your computer...)
4. mamba install xlrd h5py lxml
5. Download https://github.com/Semi-ATE/LAB-ML/tree/feature/MiniSCT-170-add-labml-toolbar-extension
6. cd LAB-ML/src)/labml_adjutancy/
7. python install -e . 
8. cd ../plugins/Semi-Control
9. python install -e .
10. call spyder, enable View->Pane 'ATE' and the View->Toolbars->ATE Plugin toolbars
11. open a project: Projects->open Project, go to LAB-ML\src\examples\project_semi_ate and select the Folder 'HATB'
12. 
[image]
    - choose production
    - set a breakpoint 
    - click the play button
    -> the window Semi-Ctrl should appear
13. 
[image]
    - select the tab 'HATB_HW0....'
    - start file in debug mode and continue
    - click the next button in 'Semi-Control' window
    - with each click on the next button the complete Testflow is running, but spyder doesn't stop at the breakpoint
    
14. I implemented a workaround:  
    - set the flag 'use breakpoints' in the window 'Semi-Control', than click next
    - spyder stops at an implemented default  'breakpoint()', if you click continue in spyder, than spyder stop at your breakpoint

I hope I haven't forgotten anything.

steff456, 2022-07-07:
Thanks for all the explanation!

nerohmot, 2022-08-23:
@steff456 any news on this subject ?

steff456, 2022-08-24:
We encounter last time that this fix needs to be done in the Spyder core side. A couple of months ago we had planned some changes for the debugger that would fix this issue, but for now it is still on pause.