Semi-ATE/Semi-ATE #265: Implement Drag & Drop function from MappingList to QTableView

hari999333, 2022-10-07:
- To insert 'No' in MappingList
- Implement Drag & Drop betwwen MappingList and QTableView
- Generate an updated '.json' file after editing the the QTableView and save it somewhere in the project directory

hari999333, 2022-10-12:
Hello, the following enhancements were made :

	1. Drag & Drop is implemented in both QTableWidget and QListWidget.
	2. The headers in the QTableWidget can be saved separately as a JSON file with their matching dictionary keys.
	3. The saved JSON file can be loaded later to get the headers in the QTableWidget.

Now, the code needs to be reviewed ...

Zlin526F, 2022-10-13:
Hello, 
please commit and push your changes in your branch '265-implement....'

hari999333, 2022-11-17:
Hello ... The enhancement mentioned in the previous comments is done with latest master in the branch "265-implement-drag-drop-function-from-mappinglist-to-qtableview-1new_branch".

Now the code needs to be reviewed. Commit ID is '9a5b05a95b29c3684a0161c9a2a9f37765529931' ...

Zlin526F, 2023-01-03:
Hello,
I used the branch 265-implement-drag-drop-function-from-mappinglist-to-qtableview-1new_branch, but it didn't run correctly.
Please check some points:

1. when I open an exel-sheet that does not have a column named 'Test name' I get an exception in line 345 from the ExcelTestWizard.py.

The Excel sheet looks like:
[image]

if you change the code like i did it will run, but I think the verify should not use mappingATEDic? :
[image]

2.  you can drop a selected test parameter anywhere, so not in the 1st line. 
[image]

3. the function verify is not correct, it use mappingATEDic, but this will not be changed if you drag and drop the parameters?

closed 2023-01-30 by nerohmot