Semi-ATE/Semi-ATE #308: Can I use this to develop any bench

lihuangroove, 2022-12-25:
I don't quite understand. Is this a general test sequence management software similar to TESTSTAND or is it aimed at your equipment

heli2src, 2022-12-26:
Semi-ate can be used for the development of testbenches and testflows. It is independent of the test system used.
You can build your own test system e.g. from NI-Instruments, Keithleys or any other instrument for which a Python driver is available. 
If the pullrequest https://github.com/Semi-ATE/Semi-ATE/pull/303 is included, there should be better documentation showing how to use Semi-ATE.

lihuangroove, 2023-01-02:
good, hhah, I am making a STDF data analysis tool, which can be divided into two types: one is used for PE analysis, and the other is ATE(similar to 93k), in which your Semi-Ate-Stdf is used.