Semi-ATE/Semi-ATE #338: file_configutation_tester_org.py and labneeds updates

duke790, 2024-05-27:
Semi-ATE\Semi-ATE\src\Plugins\semi_ate_testers\semi_ate_testers\testers\file_configuration_tester.py:

1.) imort json is not needed (used)
2.) self.hw_path should have one parent less and a path separator at the end:
[image]
3.) give a warning in case the default testerconfig file is not used:
4.) add self.logger instances
[image]
5.) correct pythonPath
[image]
6.) setattr fixes
[image]

Semi-ATE\Semi-ATE\src\ATE_spyder\ate_spyder_lab_control\labml_adjutancy\ctrl\labctrl.py:

1.) initialze self.parent to init
[image]
2.) remove case self.parent is None
[image]
3.) update setShortNames
[image]

Zlin526F, 2024-08-08:
This is done in https://github.com/Semi-ATE/Semi-ATE/tree/322-rename_test

Zlin526F, 2024-08-09:
solve

closed 2024-08-09 by Zlin526F